PLC-lang/rusty #724: thread 'main' panicked for STRINGs in linux

rarris, 2023-01-05:
```
FUNCTION main : DINT
VAR
x1 : STRING := 'abc';
x2 : STRING := 'www';
END_VAR
CONCAT(x1,x2); //example1
STRING_TO_WSTRING(x2); //example2
END_FUNCTION
```

thread 'main' panicked at 'Found ArrayValue(ArrayValue { name: "", address: 0x55e2f77f9970, is_const: false, is_const_array: false, is_const_data_array: false, is_null: false, llvm_value: "  %5 = load [2049 x i16], [2049 x i16]* %4, align 2", llvm_type: ArrayType { array_type: Type { address: 0x55e2f77799f0, llvm_type: "[2049 x i16]" } } }) but expected the IntValue variant', /usr/local/cargo/git/checkouts/inkwell-946411d814d2c9f8/9ee484b/src/values/enums.rs:286:13
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

mhasel, 2023-01-05:
This only happens when trying to build from the CLI. I was trying to reproduce this on my machine with @riederm , but unsuccessful.

mhasel, 2023-01-05:
After checking with @rarris again, we were able to build his example on his end also. Closing this as can't repro.

closed 2023-01-05 by mhasel