PLC-lang/rusty #725: Strings passed to function by ref do not match signature

99NIMI, 2023-01-05:
We have some problems generating valid IR for STRINGs passed by ref to functions
```
FUNCTION foo : DINT
VAR_INPUT {ref}
    in1 : STRING; // we expect [81 x i8]* for the parameter
...
```
generated function declaration
`define i32 @foo([81 x i8]* %0)`
```
VAR
    str_default_length : STRING := 'str'; // => [81 x i8]
    str_5 : STRING[5] := 'hello';              // => [6 x i8]
END_VAR
   func(str_default_length); // OK we will match the signature
   func(str_5); // ERROR we won't match the signature [6 x i8] != [81 x i8]
   func('abc'); // ERROR we won't match the signature [4 x i8] != [81 x i8]
```
INPUT {ref} and IN_OUT have the same behavior 

Solution 1
-
We could do the same as C
Pass arrays as pointers this would lead to following function signature
`define i32 @foo(i8* %0)`

Solution 2
-
Match the behavior for strings passed by value
allocate new var on stack with [81 x i8]
perform memcpy of the string to be passed into the new allocated var
pass the new allocated var
```
  %str = getelementptr inbounds %prog, %prog* %0, i32 0, i32 0
  %1 = alloca [81 x i8], align 1
  %2 = bitcast [81 x i8]* %1 to i8*
  call void @llvm.memset.p0i8.i64(i8* align 1 %2, i8 0, i64 ptrtoint ([81 x i8]* getelementptr ([81 x i8], [81 x i8]* null, i32 1) to i64), i1 false)
  %3 = bitcast [81 x i8]* %1 to i8*
  %4 = bitcast [81 x i8]* %str to i8*
  call void @llvm.memcpy.p0i8.p0i8.i32(i8* align 1 %3, i8* align 1 %4, i32 80, i1 false)
  %5 = load [81 x i8], [81 x i8]* %1, align 1
  %call = call i32 @str_passed_as_ptr([81 x i8] %5)
```

99NIMI, 2023-01-05:
@riederm @ghaith what do you guys think?
passing strings/arrays as pointers seems like a cleaner approach to me

ghaith, 2023-01-09:
I think for {ref} we can go with the C solution, the other solution is not really ref anymore.

closed 2023-01-30 by 99NIMI