PLC-lang/rusty #732: Negative LREAL initial values

99NIMI, 2023-01-19:
```
VAR_GLOBAL
	MIN : LREAL := -1.7976931348623157e+308;
END_VAR
```
assigning negative values exceeding the min value of REAL (-3.40282347E+38f32)
will always result in -Inf

no problem with positive values

volsa, 2023-01-27:
Not entirely sure but I think the `-inf` is considered normal behavior when exceeding the minimum value (according to quick googling); at least the generated IR seems to be identical when comparing equivalent C and ST code. There's howoever one weird edge-case which you described in your issue, i.e. 
```ST
PROGRAM main
    VAR
        F32_MIN : REAL  := -3.40282347E+38;
        F32_MAX : REAL  :=  3.40282347E+38;
        F64_MIN : LREAL := -1.7976931348623157E+308;
        F64_MAX : LREAL :=  1.7976931348623157E+308;
    END_VAR
END_PROGRAM
```
will generate the same IR except for `F64_MIN`.

C IR:
```
@F32_MIN = global float 0xC7EFFFFFE0000000, align 4
@F32_MAX = global float 0x47EFFFFFE0000000, align 4
@F64_MIN = global double 0xFFEFFFFFFFFFFFFF, align 8
@F64_MAX = global double 0x7FEFFFFFFFFFFFFF, align 8
```

Rusty IR:
```
// Values from left to right within `{...}`: F32_MIN, F32_MAX, F64_MIN, F64_MAX
@main_instance = global %main { float 0xC7EFFFFFE0000000, float 0x47EFFFFFE0000000, float 0xFFF0000000000000, double 0x7FEFFFFFFFFFFFFF }
                                                                                    ^ == F64_MIN converted to a float and invalid hex value
```

Also unrelated to this issue but casting the following
```
F64_MAX : LREAL :=  LREAL#1.7976931348623157E+308;                                          
```
yields and internal error:
```
<internal>: Unresolved constant 'F64_MAX' variable: Cannot resolve constant: LREAL#LiteralReal { value: "1.7976931348623157E+308" }
```

ghaith, 2023-01-27:
@volsa are you on the mac? is this an architecture issue?

volsa, 2023-01-27:
@ghaith Yup, currently working on the Macbook.

ghaith, 2023-01-27:
> It is, but the number used by @99NIMI is the MIN double (f64) value, it should not have been -inf there is no overflow here (did not read the full message sorry ~also when forcing the cast to LREAL it works~), so we are probably eagerly parsing a REAL (f32) at that position.

Nevermind, it looks like casting is broken

PR #742 (2023-01-31): Correctly annotate REAL and LREAL values
Fixes #732 

Edit: The casting issue is still present with this PR, see https://github.com/PLC-lang/rusty/issues/740
closed 2023-01-31 by volsa