PLC-lang/rusty #750: Array initializer must be an initializer list!

rarris, 2023-02-01:
```
PROGRAM mainProg
VAR
FB_File : file;
doData : ARRAY [0..3] OF BYTE := 0,0,0,0;
END_VAR
FB_File(
	var1 := 16#2066,
	var2 := 0,
	index1 := doData,
	index2 := 2,
	on := TRUE );
END_PROGRAM
```

```
FUNCTION_BLOCK file
VAR_INPUT
var1 : WORD;
var2 : BYTE;
index1 : ARRAY [1..4] OF BYTE;
index2 : USINT;
on : BOOL;
END_VAR
VAR_OUTPUT
END_VAR
VAR
END_VAR
;
END_FUNCTION_BLOCK
```

mainProg.st:9:2:{9:2-9:8}: error: Array initializer must be an initializer list!
mainProg.st:10:2:{10:2-10:13}: error: Expected identifier or '('
mainProg.st:11:2:{11:2-11:12}: error: Expected identifier or '('

ghaith, 2023-02-01:
might be a regression from #722

99NIMI, 2023-02-01:
https://github.com/PLC-lang/rusty/blob/dbdf32517d778fa7bcbf036664ad2fc183708179/src/validation.rs#L468-L475

yeah it slipped through with #722 
on line 475 right can also be a reference of an ARRAY

closed 2023-02-13 by 99NIMI