PLC-lang/rusty #754: overflow for STRUCT reference to itself

rarris, 2023-02-01:
```
TYPE STRUCT1 :
STRUCT
ref_to_struct : STRUCT1;
END_STRUCT
END_TYPE
```

thread 'main' has overflowed its stack

99NIMI, 2023-02-01:
it's the expected behavior
we already have validation for this #721 
this won't happen anymore after we refactored our error handling
in future we won't trigger codegen with this error

rarris, 2023-02-01:
This will occur if the standard functions are included
It seems the internal TIME->TIME error makes it so other diagnostics are not shown

volsa, 2023-02-01:
Alright, this will be fixed with PR https://github.com/PLC-lang/rusty/pull/751. I guess we'll leave it open until then?

volsa, 2023-03-27:
Can this be closed? The codegen issue is tracked in https://github.com/PLC-lang/rusty/issues/192 and multiple recursion errors are displayed.