PLC-lang/rusty #781: OSCAT Validation

99NIMI, 2023-02-17:
We can compile OSCAT but still have many validation errors.

- [ ] The type DWORD 32 is too small to hold a Pointer `This validation is OK`

- [ ] REF_TO array passing as input => Invalid assignment: cannot assign '...' to '...'

* Expression is not assignable:
    - [ ] Pointer deref assignment => ptr^ := ...
    - [ ] Pointer deref array assignment => ptr^[i] := ...
    - [ ] Bit access assignment => var.x := ...
    - [ ] Array access bit access arr[i].x := ...
    - [ ] Array access bit access(array access) arr[i].arr2[x] := ...
* Functions:
    - [ ] REAL passed DINT => Invalid assignment: cannot assign 'DINT' to 'REAL'
* Generic functions:
    - [ ] ANY_BIT passed integer => Invalid type nature for generic argument. INT is no Bit.
    - [ ] ANY_BIT passed [DWORD - DWORD => DINT] => Invalid type nature for generic argument. INT is no Bit.
    - [ ] SEL ANY passed [DINT, BYTE] => Invalid type nature for generic argument. BYTE is no Any.
    - [ ] LIMIT ANY_ELEMENTARY passed [DINT, BYTE, BYTE] => Invalid type nature for generic argument. BYTE is no Elementary.
* Variadic generic functions:
    - [ ] MIN ANY_ELEMENTARY passed [UINT, UINT - 1],[UINT, DINT] => Invalid type nature for generic argument. UINT is no Elementary.

ghaith, 2023-02-17:
The issue with assignments should be solved here : https://github.com/PLC-lang/rusty/issues/778 
I'll add tests for pointers as well

99NIMI, 2023-02-21:
The issues with generic validation should be solved with #780