PLC-lang/rusty #783: Gitpod pulls arm64 docker image, while it runs on amd64

riederm, 2023-02-18:
When running the gitpod image, it pulls `linux/arm64` and cannot run.

` ---> [Warning] The requested image's platform (linux/arm64) does not match the detected host platform (linux/amd64) and no specific platform was requested`

```console
Building the workspace image...
Sending build context to Docker daemon  113.7MB
Step 1/5 : FROM ghcr.io/plc-lang/rust-llvm:latest
latest: Pulling from plc-lang/rust-llvm
65cbd465e3f6: Pull complete 
2826bb80f013: Pull complete 
1c9c4126834a: Pull complete 
f22f9052ea07: Pull complete 
d9ad64831807: Pull complete 
cc156f0eb434: Pull complete 
0024235f2a71: Pull complete 
732c8cc59299: Pull complete 
79db2edc7737: Pull complete 
Digest: sha256:6951a0a5a83b11804e2f32b7f32bbcca1c406dfcb0015a138ffb345e95cedb10
Status: Downloaded newer image for ghcr.io/plc-lang/rust-llvm:latest
 ---> 3335bbb65781
Step 2/5 : ENV DEBIAN_FRONTEND=noninteractive
 ---> [Warning] The requested image's platform (linux/arm64) does not match the detected host platform (linux/amd64) and no specific platform was requested
 ---> Running in bc8bd0491a63
Removing intermediate container bc8bd0491a63
 ---> 1ddd92d8197a
Step 3/5 : RUN cargo install cargo-insta cargo-watch
 ---> [Warning] The requested image's platform (linux/arm64) does not match the detected host platform (linux/amd64) and no specific platform was requested
 ---> Running in 236e258b45d0
exec /bin/sh: exec format error
The command '/bin/sh -c cargo install cargo-insta cargo-watch' returned a non-zero code: 1
Image Build Failed
```

ghaith, 2023-02-19:
As a workaround set it to use this `FROM ghcr.io/plc-lang/rust-llvm:14-1.67` instead
This will however be problematic for builds as well I'll work on it asap

riederm, 2023-02-19:
> As a workaround set it to use this `FROM ghcr.io/plc-lang/rust-llvm:14-1.67` instead This will however be problematic for builds as well I'll work on it asap

no rush, i just wanted to have an issue for it

ghaith, 2023-02-19:
The images should be back to normal now, i was pushing the arm image with the same tag accidentally.
The manifest that would take care of giving the same name to both platforms was broken.
Now the images should be pulled correctly for the correct platform.
[image]
https://github.com/PLC-lang/rust-llvm-images/pkgs/container/rust-llvm/71640966?tag=latest

riederm, 2023-02-20:
it is definetely better now, but gitpod still says: `cannot find git executable, make sure it is installed as part of gitpod image: exec: "git": exec utable file not found in $PATH`

i'll try this on my local linux machine when I got time ...

ghaith, 2023-03-27:
This should now be working with the latest addition of git in the docker file

closed 2023-03-27 by ghaith