PLC-lang/rusty #792: error: relocation R_AARCH64_ABS64

rarris, 2023-02-22:
```
PROGRAM mainProg
VAR
END_VAR
VAR_INPUT
    i_enum1 : ENUM1;
END_VAR
VAR_OUTPUT
    i_struct1 : REF_TO STRUCT1;
END_VAR
CASE i_enum1 OF
     def1 : i_struct1 := ADR(g_def1);
     def2 : i_struct1 := ADR(g_def2);
     def3 : i_struct1 := ADR(g_def3);
     def4 : i_struct1 := ADR(g_def4);
     def5 : i_struct1 := ADR(g_def5);
ELSE            
END_CASE
END_PROGRAM
```

```
TYPE ENUM1 : ( def1 := 0,
               def2 := 1,
               def3 := 2,
               def4 := 3,
               def5 := 4,
               def6 := 5,
               def7 := 6,
               def8 := 7
);
END_TYPE
```

```
TYPE STRUCT1 :
STRUCT
    warning : ARRAY [0..19] OF STRUCT2;
    warningB : BOOL;
    change : DWORD;
    number : USINT;
END_STRUCT
END_TYPE
```

```
TYPE STRUCT2 :
STRUCT
    var1 : DINT;
    var2 : DINT;
    var3 : DINT;
    var4 : DINT;
END_STRUCT
END_TYPE
```

```
VAR_GLOBAL
g_def1 : STRUCT1;
g_def2 : STRUCT1;
g_def3 : STRUCT1;
g_def4 : STRUCT1;
g_def5 : STRUCT1;
END_VAR
```

Error: GeneralError { message: "lld: error: relocation R_AARCH64_ABS64 cannot be used against symbol 'g_def1'; recompile with -fPIC

the error occurs only if the CASE has more than 2 elements in the list

ghaith, 2023-03-14:
It seems this works when changing the output from shared to PIC. Will need further investigation

ghaith, 2023-08-16:
The same behaviour is reproducible in clang when compiling normal object files and linking them to shared libs.
PIC is required for linking shared libraries, the fix it to enforce it or produce an error.
When reproducing similar code in C, all compilers complain about a missing -fPIC option when producing a shared library.
I am not  sure why the linker doesn't complain in our cases, but it seems like the fix would be to use pic with shared libraries and allow the DynamicNoPIC option only when specifically requested (--shared always does pic, and a new --no-pic option enforces a no pic behaviour where linking is not guaranteed to work)

PR #937 (2023-08-17): Make shared and PIC the same, add no-pic option
Also fixed an issue with the default name overwriting the input, the output name is now <input>.out if not specified

Fixes #792 and #936
closed 2023-08-17 by ghaith