PLC-lang/rusty #794: Could not resolve reference ...

rarris, 2023-02-27:
```
PROGRAM mainProg
VAR_INPUT
	Input : STRUCT3;
END_VAR
VAR_OUTPUT
	Output : STRUCT1;
END_VAR
VAR
END_VAR
Output.struct1_var.struct2_nr1 := Input.struct3_var; 
END_PROGRAM
```

```
TYPE STRUCT1 :
STRUCT
struct1_var : DWORD;
END_STRUCT
END_TYPE
```

```
TYPE STRUCT2 :
(
struct2_nr1 := 1,
struct2_nr2 := 2
);
END_TYPE
```

```
TYPE STRUCT3 :
STRUCT
struct3_var : BOOL;
END_STRUCT
END_TYPE
```

mainProg.st:11:20:{11:20-11:31}: error: Could not resolve reference to struct2_nr1
mainProg.st:11:1:{11:1-11:31}: error: Expression is not assignable
Error: SyntaxError { message: "no type hint available for Reference {\n    name: \"struct2_nr1\",\n}", range: [SourceRange { range: 133..144,

riederm, 2023-02-28:
and @riederm also pointed out that this whole bit-access-concept is a nasty thing! 
we should make PLC-Developer do a special syntax highlighting for bit-access. Its so easy to miss.

ghaith, 2023-02-28:
Enum/vaiable access is implemented with %X<varName>

riederm, 2023-03-01:
ok ... so we should discuss the following:
- should we try to link against a constant/enum if a qualified reference does not resolve
- and in addition validate against this with a diagnostic like: `Could not resolve reference to 'varName': If you meant to access a bit use '%XvarName' instead.`

closed 2023-07-23 by mhasel