PLC-lang/rusty #796: enum declaration in VAR block

rarris, 2023-02-28:
example:
```
VAR
var1 : (x1:=1, x2:=2, x3:=3) := stop;
END_VAR
```

codesys support this
should we support also?

riederm, 2023-02-28:
we have a test `inline_enums_are_generated` which checks for that - so in general we can handle enums in declaration blocks.
where exactly is the issue?  I did some experiments in unit-tests and at least I get some code generated - I dont know if it really runs, I didnt test this far.

can yo add the outputs that you get?
I tried this code in a unit test:
```st
        VAR_GLOBAL
          x : (red, yellow, green) := 2;
        END_VAR

        PROGRAM  xxx
            VAR
                y : (redy:=1, yellowy:=2, greeny:=3) := 2;
            END_VAR
            VAR
                var1 : (x1:=1, x2:=2, x3:=3) := stop;
                //or
                var2 : (x5, x6, x7) := stop2;
            END_VAR
        END_PROGRAM
```

btw. I think the initializer is ignored in the godegen!

ghaith, 2023-02-28:
We don't validate in unit tests so you might have missed them. But it seems like stop and stop2 are not declared variables.

mhasel, 2023-02-28:
Aren't our enums constant? Allowing assignments to enums outside of their declaration just doesn't sit well with me

ghaith, 2023-02-28:
> Aren't our enums constant? Allowing assignments to enums outside of their declaration just doesn't sit well with me

The enums are, the variables containing an enum that can be assigned to a value of the enum.
In Codesys this value can be also outside the range of the enum, usually with a warning in PLC-Developer

mhasel, 2023-02-28:
Yeah I just pieced it together that it might look something like this, which seems reasonable.
`enum1 : (start:=1, stop:=2) := stop`

rarris, 2023-03-01:
> 
@riederm 

error: Unexpected token: expected KeywordSemicolon but found ':= 2'
   ┌─ test.st:65:54
   │
65 │                 y : (redy:=1, yellowy:=2, greeny:=3) := 2;
   │                                                                             ^^^^ Unexpected token: expected KeywordSemicolon but found ':= 2'

error: Unexpected token: expected KeywordSemicolon but found ':= stop'
   ┌─ test.st:66:46
   │
66 │                 var1 : (x1:=1, x2:=2, x3:=3) := stop;
   │                                                             ^^^^^^^ Unexpected token: expected KeywordSemicolon but found ':= stop'

error: Unexpected token: expected KeywordSemicolon but found ':= stop2'
   ┌─ test.st:68:37
   │
68 │                 var2 : (x5, x6, x7) := stop2;
   │                                            ^^^^^^^^ Unexpected token: expected KeywordSemicolon but found ':= stop2'

closed 2023-07-23 by mhasel