PLC-lang/rusty #822: cannot assign 'DINT' to 'STRING'

rarris, 2023-03-23:
```
PROGRAM  main
VAR
a : STRING[80];
b : STRUCT1;
END_VAR
a := UINT_TO_STRING(b.var10);    //var10 is UINT
END_PROGRAM
```

error: Invalid assignment: cannot assign 'DINT' to 'STRING'

ghaith, 2023-05-08:
The function UINT_TO_STRING is not yet implemented in the standard functions, but the signature here is wrong, and therefore you get an error, the signature should be 
```
FUNCTION UINT_TO_STRING : STRING
VAR_INPUT
    in : UINT;
END_VAR
END_FUNCTION
```

rarris, 2024-03-06:
we have a new error case here

code example:
```
PROGRAM mainProg
VAR
  stfConv : STRING[125];
  bytIndex : USINT;
  Args : argsSTR;
END_VAR
  stfConv := BOOL_TO_STRING(Args.b[bytIndex]);
END_PROGRAM

TYPE argsSTR :
	 STRUCT
		b : ARRAY[0..4] OF BOOL ;
		i : ARRAY[0..4] OF DINT ;
	END_STRUCT
END_TYPE
```
error: Invalid assignment: cannot assign 'DINT' to 'STRING[125]'
Error: E001: Compilation aborted due to critical errors

volsa, 2025-08-07:
Not sure, a `TO_STRING__BOOL` is missing and could be the issue? See the `to_string.st` file, which hash a signature of `FUNCTION TO_STRING<T: ANY> : STRING[__STRING_LENGTH]` but no corresponding implementation. Weird issue nevertheless

volsa, 2026-09-29:
The first example fails because of the user-declared signature `UINT_TO_STRING : DINT`, so that error is correct. The standard library now provides the TO_STRING conversions, including `BOOL_TO_STRING` added in 44b84d619bb (#1903), and both examples compile and run correctly with the stdlib on master.

closed 2026-09-29 by volsa