PLC-lang/rusty #828: Wrong downcast warning

volsa, 2023-03-27:
The following code
```Smalltalk
FUNCTION main : INT
    VAR
        a, b, c : INT;
    END_VAR

    a := b + c;
    a := a * b + c;
END_FUNCTION
```
will report
```Bash
warning: Potential loss of information due to assigning 'DINT' to variable of type 'INT'.
  ┌─ demo.st:6:5
  │
6 │     a := b + c;
  │     ^^^^^^^^^^ Potential loss of information due to assigning 'DINT' to variable of type 'INT'.

warning: Potential loss of information due to assigning 'DINT' to variable of type 'INT'.
  ┌─ demo.st:7:5
  │
7 │     a := a * b + c;
  │     ^^^^^^^^^^^^^^ Potential loss of information due to assigning 'DINT' to variable of type 'INT'.
```
which is wrong because `a`, `b` and `c` are of type INT

99NIMI, 2023-03-29:
https://github.com/PLC-lang/rusty/blob/47bb28f1244ea38bfeb892d7fdc230697f0c13d7/src/resolver.rs#L897-L911

i think we can replace everything in the first else with 
`get_bigger_type(left_type, right_type, self.index)`
it should do the job for bit types and i'm not sure we why we called `get_bigger_type(..)` a second time with DINT

ghaith, 2023-03-31:
Default behavior on all arithmetic operations is to do them with DINT (same as C does with the int type)
All literals are also DINT based, so a := a + 100 will also have the same issue
 
The warning you get is _technically_ correct, the addition and multiplication will always default to DINT.

Not sure yet how we solve this, @99NIMI's idea is certainly a possibility if there is no reason for the conversion. Another is we need to mark this type of conversions (where all types are the same) as legal.. 
@riederm  ideas? TBH I can't recall the reason we followed the C convention here. 
I know we wanted to have the same datatype everywhere in the operation, but not why we forced DINT (initially we forced INT actually) Was it performance? (i32 operations are easier)

PR #1114 (2024-03-18): fix: false-positive downcast warnings
Fixes false-positive downcast warnings on integrally promoted expressions. This is achieved by checking each element's type in the expression individually and comparing it to the resulting type rather than checking the expression's type as a whole and mirrors the behaviour of CodeSys.

resolves #828
closed 2024-03-18 by mhasel