PLC-lang/rusty #830: Project structure

volsa, 2023-03-28:
Tracking issue for [cargo workspaces](https://doc.rust-lang.org/book/ch14-03-cargo-workspaces.html), which we've already discussed internally. As a starting point something like seems ok?
```
rusty/
rusty-ast/
rusty-parser/
rusty-resolver/
rusty-validator/
rusty-codegen/
```
This will bloat up the repository itself, so maybe we can pack these workspaces into a `crates` folder, such that the current structure of the project looks like:
```
book/
crates/
examples/
scripts/
README.md
...
```
Feedback welcome :)

ghaith, 2023-03-31:
the repo is called rusty, so can drop the name from the subsequent folders maybe? 
I like the crates idea, and the top level would initially include things that have not been moved yet.
Eventually it would be empty

ghaith, 2023-03-31:
One more thing, the first workspace element is the Standard functions i guess :) there is an issue for that.

volsa, 2023-03-31:
> the repo is called rusty, so can drop the name from the subsequent folders maybe?

I think the naming prefix is the de facto standard, i.e. many popular projects use them; see
- https://github.com/rust-lang/rust/tree/master/compiler
- https://github.com/tauri-apps/tauri/tree/dev/core
- https://github.com/alacritty/alacritty
- https://github.com/helix-editor/helix

> One more thing, the first workspace element is the Standard functions i guess :) there is an issue for that.

Thanks, totally forgot them. This should be the corresponding issue https://github.com/PLC-lang/StandardFunctions/issues/80