PLC-lang/rusty #860: 2-complement signed hex number literals (INT#FFFF == -1?)

riederm, 2023-05-13:
Should rusty be able to reprsent negative hex numbers as literals?
#841 will not allow a line like this:
`x : INT := INT16#FFFF:` 
in terms of the bit representation, `1111 1111 1111 1111` it is -1 since the declared type is a signed 16 bit integer. I guess it is the same for 
`x: INT := INT2#1111_1111_1111_1111;`

volsa, 2023-09-14:
I think this is a fundamental parser issue, i.e. any cast statement such as `INT#16#FFFF` is cast into a `i128` value by the parser[0]. Once inside the const-evaluator module, overflow checks happen with regard to the cast type (here `INT` or `i16`) but `FFFF` being parsed into a `i128` value, we get `65535` which is an overflow since it exceeds `i16::MAX`[1]. If correct, do we just report these kind of overflows in the parser?

[0] <https://github.com/PLC-lang/rusty/blob/84b859d7c354a3f7f6a3114d455e0e55b7e87189/src/parser/expressions_parser.rs#L496>

[1] https://github.com/PLC-lang/rusty/blob/84b859d7c354a3f7f6a3114d455e0e55b7e87189/src/resolver/const_evaluator.rs#L265

mhasel, 2024-03-13:
I've looked into this a bit and also tested the behaviour of our old system, CodeSys 3 and Rust. Neither allow 2s complement literals to be assigned to signed types if the value overflows.
To implement this, we would probably need to change our `AstLiteral::Integer` to carry metadata about the radix, 
```
pub enum IntegerValue {
    Binary(i128),
    Octal(i128),
    Decimal(i128),
    Hexadecimal(i128)
}
```
which will necessitate a lot of changes throughout the compiler. This impact could be lessened a bit by implementing traits like `Deref` etc. on the enum, but I'm not sure if it's worth it. Personally I am leaning towards a won't fix and suggesting a cast to an unsigned value before casting to the signed variant, like Rust does:
```
to use as a negative number (decimal `-1`), consider using the type `u16` for the literal and cast it to `i16`: `0xFFFFu16 as i16`
```
On the other hand, to be able to accurately suggest this we will need metadata on the integer values too.
Also, explicit casts like `INT#UINT#16#FFFF` are currently not possible (see #1151).