PLC-lang/rusty #867: memset and memcpy require linking against libc

ghaith, 2023-06-02:
**Describe the bug**
If we generate an application that uses memcpy or memset (i.e. using strings or assigning full structs and arrays)
it needs to be linked against a libc. Previously the standard functions used to link explicitly against libc, this is no longer the case.
I suggest we re-introduce this option either in the stdlib or on all compiled applications that use memcpy and memset.

We could look into statically linking [musl](https://www.musl-libc.org/) or [newlib](https://sourceware.org/newlib/) or simply dynamically linking whatever version is on the system.

Alternatively we could fail compilation with a message so that the user adds `-lc` or link against their preferred `libc`

corbanvilla, 2023-07-26:
No worries at all @ghaith.

What build step are you referring to? It looks like the shared library already includes the math library [here](https://github.com/PLC-lang/rusty/blob/3fa491681eac3bb2f9af379dca383be27907c432/scripts/build.sh#L238). Is that where you were wanting to also include `-lc`?

I ran into some issues building a static binary that links stdlib, so ended up going the musl route with [musl.cc](http://musl.cc/). Happy to work on integrating it further if it would be helpful.

ghaith, 2023-07-31:
> No worries at all @ghaith.
> 
> What build step are you referring to? It looks like the shared library already includes the math library [here](https://github.com/PLC-lang/rusty/blob/3fa491681eac3bb2f9af379dca383be27907c432/scripts/build.sh#L238). Is that where you were wanting to also include `-lc`?

Yes It would be there.
> 
> I ran into some issues building a static binary that links stdlib, so ended up going the musl route with [musl.cc](http://musl.cc/). Happy to work on integrating it further if it would be helpful.

Yes static linking stdlib is usually not possible, musl support would be nice indeed. We need to see how this fits in general, if we make it an option while building or leave it to the user. If we do leave it to the user, we will need to add the option in the build.sh file to allow for musl and compile the standard lib with musl as a target.

I'll start a discussion for that.

volsa, 2026-09-29:
Since ef83a8adb67 (#1638) the default linker is the `cc` driver (falling back to clang, then ld.lld), which links libc implicitly, so programs using memcpy/memset link and run without `-lc` (`--nolibc` opts out). The stdlib shared object also lists libc.so.6 as NEEDED. musl support is tracked in #924.

closed 2026-09-29 by volsa