PLC-lang/rusty #897: `--help` panics

volsa, 2023-07-11:
Running `./target/release/plc --help` yields the following panic
```
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: GeneralError { message: "plc_driver 0.1.0\nIEC61131-3 Structured Text compiler powered by Rust & LLVM\n\nUSAGE:\n    plc [OPTIONS] <input-files>...\n    plc [OPTIONS] <SUBCOMMAND>\n\nARGS:\n    <input-files>...    Read input from <input-files>, may be a glob expression like 'src/**/*'\n                        or a sequence of files\n\nOPTIONS:\n        --bc\n            Emit binary IR (binary representation of LLVM-IR) as output\n\n    -c\n            Do not link after compiling object code\n\n        --check\n
Check only, do not generate any output\n\n        --debug-variables\n            Generate debug information for global variables\n\n        --encoding <encoding>\n            The file encoding used to read the input-files, as defined by the Encoding Standard\n\n        --error-format <error-format>\n            Set format for error reporting [default: rich] [possible values: rich, clang, none]\n\n    -g, --debug\n            Generate source-level debug information\n\n    -h, --help\n            Print help information\n\n        --hardware-conf <hardware-conf>\n            Generate Hardware configuration files to the given location.\n                Format is detected by extenstion.\n                Supported formats : json, toml\n\n    -i, --include <include>\n            Include source files for external functions\n\n        --ir\n            Emit IR (LLVM Intermediate Representation) as output\n\n    -j, --threads <threads>\n            Set the number of threads to use for the compilation\n\n    -l, --library <library>\n            Library name to link\n\n    -L, --library-path <library-path>\n            Search path for libraries, used for linking\n\n        --linker <linker>\n            Define a custom (cc compatible) linker command\n\n    -o, --output <output-file>\n            Write output to <output-file>\n\n    -O, --optimization <optimization>\n            Optimization level [default: default] [possible values: none, less, default, aggressive]\n\n        --pic\n            Equivalent to --shared\n\n        --relocatable\n            E
```

Starsmith2, 2023-07-13:
Same panic in "debug/plc"(master branch).

riederm, 2023-07-18:
i have no idea why I would remove the gfi label???

closed 2023-09-05 by volsa