PLC-lang/rusty #918: bit-access on pointers will panic without error/validation

mhasel, 2023-07-21:
We are missing validations for the following examples, all of which end either in an unreachable or panic.

For regular pointer variables, I don't think bit-access makes much sense, but what about structs etc.?
In some ST dialects/implementations it is possible to directly bit-access into a struct, but I guess the structs in these variants are packed? Should we also support this or just return an error and abort codegen?

```
TYPE struct_t : STRUCT
	field : DINT;
END_STRUCT END_TYPE

FUNCTION_BLOCK fb_t
END_FUNCTION_BLOCK

PROGRAM mainProg
	VAR
		st : struct_t;
		fb : fb_t;
		ptr : REF_TO DINT;
	END_VAR
		// None of these are currently validated
		st.%X1;   // hits unreachable in cast_if_needed
		fb.%X1;   // hits unreachable in cast_if_needed
		ptr.%X1;  // panics in Inkwell
END_PROGRAM
```

volsa, 2024-04-02:
Has this been fixed? I'm unable to reproduce on the current master

mhasel, 2024-04-02:
Looks like a validation for this has been added with #951. That fixes the issue of hitting unreachables/panics, but we are still unable to directly bit-access into structs, so I guess this is now more of a discussion of whether or not we want to support that functionality. 

Updated example:
```
TYPE struct_t : STRUCT
	field : DINT;
END_STRUCT END_TYPE

PROGRAM mainProg
	VAR
		st : struct_t := (field := 2);
	END_VAR
		st.%X1; // should this return the second bit of `st.field`, i.e. 1?
END_PROGRAM
```

It might make sense to close this issue and open a new one for this use-case?

volsa, 2026-09-29:
All three cases from the example now give validation errors instead of a panic: E055 for `st.%X1` and `fb.%X1`, and E141 ("dereference with ^ first") for `ptr.%X1`. The validation came with 3c36dd763ff (#951), and the pointer message was refined later. Direct bit-access into structs can be a separate feature request, as suggested above.

closed 2026-09-29 by volsa