PLC-lang/rusty #920: Unable to generate IR for `string_functions.st`

corbanvilla, 2023-07-21:
**Describe the bug**
I'm attempting to generate the LLVM IR for the stdlib, but `string_functions.st` causes an issue. Specifically, it appears that [lines 560-685](https://github.com/PLC-lang/rusty/blob/e3af00b9b6f04d66e20bacd97da20a2e31ef124c/libs/stdlib/iec61131-st/string_functions.st#L560-L685) are causing the problem.

**To Reproduce**
Steps to reproduce the behavior:
```bash
> cargo build
> ./target/debug/plc --ir ./libs/stdlib/iec61131-st/string_functions.st 

"/tmp/.tmp3DPFle/libs/stdlib/iec61131-st/string_functions.ll:304:9: error: stored value and pointer type do not match\n  store i8* %2, [81 x i8]*** %5, align 8\n        ^\n"
```

**Expected behavior**
Generate valid IR and not error out. Once these lines are disabled, the IR for this file and all others generates without a problem.

I've attempted this on the latest build as well as the last few commits, but they all have been yielding this issue.

Any help is very much appreciated!

ghaith, 2023-07-24:
@mhasel a good old friend returns. 

@corbanvilla could yo try this on the 0.2 release please? It might be the parallel build that broke this.

corbanvilla, 2023-07-26:
@ghaith Yes, it works fine for 0.2 release.

Also tested with `RUST_BACKTRACE=1` but no additional information is included.

PR #926 (2023-08-04): fix `string_functions.st` error when compiled to IR
fixes #920

String equality comparison functions were missing resolved-type header implementations (e.g. GT__STRING) with the proper variadic signature.
closed 2023-08-04 by mhasel