PLC-lang/rusty #932: ARRAY in ARRAY initialization error

rarris, 2023-08-10:
```
PROGRAM mainProg
VAR
END_VAR
;
END_PROGRAM
```

```
TYPE STRUCT1 :
 STRUCT
  Node : ARRAY[0..15] OF 
   STRUCT2 := 
   [
   (
    var1_struct2 := 'text',
    enum_struct2 := enum_var1,
    arr_struct2 := 
    (
     var1 := 'Type',
     var2 := 3,
     var3 := 2
    )
   )
   ]
  ;
 END_STRUCT
END_TYPE
```

```
TYPE STRUCT2 :
	 	STRUCT
		var1_struct2 : STRING[31];
		enum_struct2 : ENUM1;
		var3_struct2 : UINT;
		arr_struct2 : ARRAY[0..19] OF STRUCT3;
	END_STRUCT
END_TYPE
```

```
TYPE ENUM1 :
		( 
		enum_var1,
		enum_var2,
		enum_var3,
		enum_var4 
	);
END_TYPE
```

**err:**
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: CombinedDiagnostic { message: "Some initial values were not generated", inner_diagnostics: [SyntaxError { message: "Cannot generate literal initializer for 'STRUCT1.Node': Value cannot be derived", range: [SourceRange { range: 79..349, file: Some("...\\src\\STRUCT1.dt") }, SourceRange { range: 5..12, file: Some("...\\src\\STRUCT1.dt") }], err_no: codegen__general }], err_no: codegen__general }', compiler\plc_driver\./src/main.rs:7:32
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

volsa, 2023-08-10:
This looks somewhat similar to https://github.com/PLC-lang/rusty/issues/707, let's re-check if solve it also solves this issue.

volsa, 2023-08-22:
This will be fixed with https://github.com/PLC-lang/rusty/pull/938 although validations seem to be missing to get a clear answer why the codegen failed here. Specifically this assignment `arr_struct2 := (var1 := 'Type', var2 := 3, var3 := 2)` has to start with a leading `[` symbol, i.e. `arr_struct2 := [(var1 := 'Type', var2 := 3, var3 := 2)]`. Can you re-check this just to be sure @rarris ?

volsa, 2023-11-21:
I'm closing this in favor of https://github.com/PLC-lang/rusty/issues/870 as the codegen issue seems to be fixed

closed 2023-11-21 by volsa