PLC-lang/rusty #939: Warn when assignment makes no sense

volsa, 2023-08-17:
I just spent a good hour trying to debug a failing correctness test before realising the code is perfectly fine if not for the assignment. Assume you have the following code, where you try to assign `1` to a `value` variable.
```
FUNCTION main : DINT
    VAR
        value : DINT := 0;
    END_VAR

    value = 1;
END_FUNCTION
```
The assignment will never happen because `=` should have been `:=`. I feel like these kind of "typos" are hard to find, hence some warning / suggestion for such cases where a user meant to use an assignment rather than a comparison are (imo) a huge QOL change.

riederm, 2023-11-09:
we should have a warning-validation for "this statement has no effect".
does it actually boil down to r-values?

volsa, 2023-11-09:
I guess? The easiest way I assume would be to check if binary expressions with an equal operator have a type-hint on `bool` and if not slap a warning on it.

closed 2023-12-19 by volsa