PLC-lang/rusty #955: Windows build fail: Link error

VasenevEA, 2023-08-31:
**System info**

`Windows 10.0.19045`
`LLD 14.0.6`
`cargo 1.72.0 (103a7ff2e 2023-08-15)`

**Reproduce Command**
`cargo build`
or
`cargo build -p plc_driver`

**Output**

```
...
Compiling plc_project v0.1.0 (E:\rusty\compiler\plc_project)
Compiling plc_xml v0.1.0 (E:\rusty\compiler\plc_xml)
Compiling plc_driver v0.1.0 (E:\rusty\compiler\plc_driver)
error: linking with `lld-link` failed: exit code: 1

...

note: lld-link: error: undefined symbol: __std_init_once_link_alternate_names_and_abort
          >>> referenced by libllvm_sys-f4356f940a2a0df6.rlib(X86TargetMachine.obj):($LN51)
          >>> referenced by libllvm_sys-f4356f940a2a0df6.rlib(X86TargetMachine.obj):($LN58)
          >>> referenced by libllvm_sys-f4356f940a2a0df6.rlib(X86TargetMachine.obj):($LN18)
          >>> referenced 1150 more times

          lld-link: error: undefined symbol: __std_find_trivial_1
          >>> referenced by libllvm_sys-f4356f940a2a0df6.rlib(MipsTargetMachine.obj):($LN35)
          >>> referenced by libllvm_sys-f4356f940a2a0df6.rlib(MipsTargetMachine.obj):($LN33)
          >>> referenced by libllvm_sys-f4356f940a2a0df6.rlib(MipsTargetMachine.obj):($LN794)
          >>> referenced 66 more times

          lld-link: error: undefined symbol: __std_find_trivial_8
          >>> referenced by libllvm_sys-f4356f940a2a0df6.rlib(AMDGPUAsmPrinter.obj):($LN37)
          >>> referenced by libllvm_sys-f4356f940a2a0df6.rlib(AMDGPUAsmPrinter.obj):($LN87)
          >>> referenced by libllvm_sys-f4356f940a2a0df6.rlib(AMDGPUAsmPrinter.obj):(void const ** __cdecl __std_find_trivial<void const *, void const *>(void const **, void const **, void const *const))
          >>> referenced 1009 more times

          lld-link: error: undefined symbol: __std_find_trivial_4
          >>> referenced by libllvm_sys-f4356f940a2a0df6.rlib(InstructionCombining.obj):($LN1952)
          >>> referenced by libllvm_sys-f4356f940a2a0df6.rlib(InstructionCombining.obj):($LN1952)
          >>> referenced by libllvm_sys-f4356f940a2a0df6.rlib(InstructionCombining.obj):($LN31)
          >>> referenced 132 more times

          lld-link: error: undefined symbol: __std_max_element_1
          >>> referenced by liblld_rs-9cda65437188f5f9.rlib(DataLayout.obj):($LN28)
          >>> referenced by liblld_rs-9cda65437188f5f9.rlib(DataLayout.obj):(unsigned char const * __cdecl __std_max_element<unsigned char const>(unsigned char const *, unsigned char const *))
          >>> referenced by libllvm_sys-f4356f940a2a0df6.rlib(SimplifyLibCalls.obj):($LN653)

          lld-link: error: undefined symbol: __std_find_trivial_2
          >>> referenced by libllvm_sys-f4356f940a2a0df6.rlib(TargetRegisterInfo.obj):($LN423)
          >>> referenced by libllvm_sys-f4356f940a2a0df6.rlib(TargetRegisterInfo.obj):($LN33)
          >>> referenced by libllvm_sys-f4356f940a2a0df6.rlib(TargetRegisterInfo.obj):(unsigned short const * __cdecl __std_find_trivial<unsigned short const, unsigned int>(unsigned short const *, unsigned short const *, unsigned int))
          >>> referenced 11 more times

          lld-link: error: undefined symbol: __std_max_element_4
          >>> referenced by libllvm_sys-f4356f940a2a0df6.rlib(SIMachineScheduler.obj):($LN934)
          >>> referenced by libllvm_sys-f4356f940a2a0df6.rlib(SIMachineScheduler.obj):($LN934)
          >>> referenced by libllvm_sys-f4356f940a2a0df6.rlib(SIMachineScheduler.obj):(int * __cdecl __std_max_element<int>(int *, int *))
          >>> referenced 3 more times

          lld-link: error: undefined symbol: __std_max_element_8
          >>> referenced by libllvm_sys-f4356f940a2a0df6.rlib(JumpThreading.obj):($LN496)
          >>> referenced by libllvm_sys-f4356f940a2a0df6.rlib(JumpThreading.obj):(unsigned __int64 * __cdecl __std_max_element<unsigned __int64>(unsigned __int64 *, unsigned __int64 *))          >>> referenced by libllvm_sys-f4356f940a2a0df6.rlib(SimplifyCFG.obj):(void __cdecl FitWeights(class llvm::MutableArrayRef<unsigned __int64>))

          lld-link: error: undefined symbol: __std_count_trivial_4
          >>> referenced by libllvm_sys-f4356f940a2a0df6.rlib(X86ISelLowering.obj):(public: __cdecl <lambda_607a4e83040d3e773181e854ebd44aee>::operator()(class llvm::ArrayRef<int>, class llvm::ArrayRef<int>, class llvm::ArrayRef<int>, class llvm::ArrayRef<int>, int, int) const)
          >>> referenced by libllvm_sys-f4356f940a2a0df6.rlib(X86ISelLowering.obj):(public: __cdecl <lambda_607a4e83040d3e773181e854ebd44aee>::operator()(class llvm::ArrayRef<int>, class llvm::ArrayRef<int>, class llvm::ArrayRef<int>, class llvm::ArrayRef<int>, int, int) const)
          >>> referenced by libllvm_sys-f4356f940a2a0df6.rlib(X86ISelLowering.obj):(public: __cdecl <lambda_607a4e83040d3e773181e854ebd44aee>::operator()(class llvm::ArrayRef<int>, class llvm::ArrayRef<int>, class llvm::ArrayRef<int>, class llvm::ArrayRef<int>, int, int) const)
          >>> referenced 3 more times

          lld-link: error: undefined symbol: __std_count_trivial_8
          >>> referenced by libllvm_sys-f4356f940a2a0df6.rlib(SimplifyCFG.obj):(bool __cdecl ForwardSwitchConditionToPHI(class llvm::SwitchInst *))
          >>> referenced by libllvm_sys-f4356f940a2a0df6.rlib(SimplifyCFG.obj):(??$count@V?$iterator_range@PEAPEAVBasicBlock@llvm@@@llvm@@PEAVBasicBlock@2@@llvm@@YA?A_P$$QEAV?$iterator_range@PEAPEAVBasicBlock@llvm@@@0@AEBQEAVBasicBlock@0@@Z)
          >>> referenced by libllvm_sys-f4356f940a2a0df6.rlib(SimplifyCFG.obj):(unsigned __int64 __cdecl __std_count_trivial<class llvm::BasicBlock *, class llvm::BasicBlock *>(class llvm::BasicBlock **, class llvm::BasicBlock **, class llvm::BasicBlock *const))
          >>> referenced 1 more times

error: could not compile `plc_driver` (bin "plc_driver") due to previous error
```

ghaith, 2023-09-01:
Hello @VasenevEA , thank you for the bug report.
What version of LLVM are you using? I see you have `LLD 14.0.6` but is it downloaded directly from the llvm page? 
LLVM typically does not ship the development libraries and headers with their windows distribution which is why we use our own custom built version (https://github.com/PLC-lang/llvm-package-windows).

volsa, 2023-09-04:
I'm unable to reproduce with the following environment
- `stable-x86_64-pc-windows-msvc (default)`, `rustc 1.72.0 (5680fa18f 2023-08-23)` (via `rustup show`)
- [LLVM 14.0.6](https://github.com/PLC-lang/llvm-package-windows/releases/tag/v14.0.6) (the `bin/` directory needs to be added to the environment variables)
- Windows 10 SDK & MSVC v142 - VS 2019 C++ x64/x86 build tools via Visual Studio Installer 

I did however need to remove the following lines for a successful `cargo test` run https://github.com/PLC-lang/rusty/blob/7d17dc62f6a95da983cbd7b8de1f20ab2f099a37/.cargo/config.toml#L19-L21 

Could you re-check if you're able to solve this issue with the given environment @VasenevEA?

VasenevEA, 2023-09-04:
> Hello @VasenevEA , thank you for the bug report. What version of LLVM are you using? I see you have `LLD 14.0.6` but is it downloaded directly from the llvm page? LLVM typically does not ship the development libraries and headers with their windows distribution which is why we use our own custom built version (https://github.com/PLC-lang/llvm-package-windows). We build on windows daily using our github workflows (
> 
> https://github.com/PLC-lang/rusty/blob/3c36dd763ffd370adbcb0e48d7825c5a8367c074/.github/workflows/rust.yml#L63
> 
> ) maybe see if anything from this configuration is missing for you?.

I've solved the problem, thanks for the help!

closed 2023-09-04 by volsa