PLC-lang/rusty #961: Skipped validation for type initializers

riederm, 2023-09-06:
It looks like some validations are skipped because of another error.

https://github.com/PLC-lang/rusty/blob/63bf5db6847154a6617a580fdb310bc2082fb7e9/src/codegen/tests/initialization_test/type_initializers.rs#L249-L265

This test only reports the error for `MyStruct2.b`
https://github.com/PLC-lang/rusty/blob/63bf5db6847154a6617a580fdb310bc2082fb7e9/src/codegen/tests/initialization_test/type_initializers.rs#L256 but not the other error(s) in line https://github.com/PLC-lang/rusty/blob/63bf5db6847154a6617a580fdb310bc2082fb7e9/src/codegen/tests/initialization_test/type_initializers.rs#L251

mhasel, 2023-09-14:
This looks to be closely related #870

volsa, 2026-09-29:
On current master the real compile pipeline validates the initializers and reports both errors from this example: `Could not resolve reference to c` for the global `a` (line 2) and for the field `MyStruct2.b` (line 8). The cited unit test is a codegen-only test that skips validation by design, so it still shows only the codegen error. Closing; please reopen if a validation is still skipped.

closed 2026-09-29 by volsa