PLC-lang/rusty #965: Add a test to show how nested array initialization with structs behaves

ghaith, 2023-09-08:
When initializing arrays of structs the initialization sequence can be confusing.

Just to make sure we have it correct, let's add a test for the following scenario

```
TYPE STRUCT1 : STRUCT
  node: DINT;
  myArr: ARRAY[0..1] OF STRUCT2;
END_STRUCT
END_TYPE

TYPE STRUCT2 : STRUCT
x1 : INT;
x2: INT;
END_STRUCT
END_TYPE

PROGRAM main
VAR
var_init1 : ARRAY[0..1] OF STRUCT1 := [node:=1, myArr:=[(x1:=1),(x2:=2)]];
var_init2 : ARRAY[0..1] OF STRUCT1 := [(node:=2, myArr:=[(x1:=3,x2:=4)])];
END_VAR
END_PROGRAM
```

The result should be:
```json
{
   "var_init1":[
      {
         "node":"1",
         "myArr":[
            {
               "x1":"0",
               "x2":"0"
            },
            {
               "x1":"0",
               "x2":"0"
            }
         ]
      },
      {
         "node":"0",
         "myArr":[
            {
               "x1":"1",
               "x2":"0"
            },
            {
               "x1":"0",
               "x2":"2"
            }
         ]
      }
   ],
   "var_init2":[
      {
         "node":"2",
         "myArr":[
            {
               "x1":"3",
               "x2":"4"
            },
            {
               "x1":"0",
               "x2":"0"
            }
         ]
      },
      {
         "node":"0",
         "myArr":[
            {
               "x1":"0",
               "x2":"0"
            },
            {
               "x1":"0",
               "x2":"0"
            }
         ]
      }
   ]
}
```

ghaith, 2023-09-08:
@rarris mentioned that it might be also good to cover these types of cases: 
```
var_init3 : ARRAY[0..1] OF STRUCT1 := [node:=2, myArr:=[x1:=3,x2:=4]];
var_init4 : ARRAY[0..1] OF STRUCT1 := [(node:=2, myArr:=[x1:=3,x2:=4])];
```

var_init3 will have all structs filled like var_init1 previously
var_ini4 will have the the first STRUCT1 filled in the array, but inside it myArr will have both structs filled.

volsa, 2023-09-08:
`var_init1 : ARRAY[0..1] OF STRUCT1 := [node:=1, myArr:=[(x1:=1),(x2:=2)]]` resulting in 
```json
"var_init1": [
    { "node":"1", "myArr":[ { "x1":"0", "x2":"0" }, { "x1":"0", "x2":"0" } ] },
    { "node":"0", "myArr":[ { "x1":"1", "x2":"0" }, { "x1":"0", "x2":"2" } ] },
]
```
feels weird to me. Is this actually defined in the norm like that (could find anything)? If not just my two cents but this feels much cleaner `... := [(node := 1), (node := 0, myArr := [(x1 := 1), (x2 := 2)])]`, i.e. where array elements have to be explicitly wrapped inside parentheses.

ghaith, 2023-09-09:
parans are of course cleaner. But we should still have it defined in tests what the strange combinations I think. Since these are valid cases syntactically. They just are not intuitive.

ghaith, 2023-09-09:
or we could enforce the the parans...

volsa, 2023-09-09:
Sorry, yeah I meant enforcing it. Just slap an error on these weird cases and "force" users to use parentheses making life easier for them and us.

closed 2023-11-13 by volsa