PLC-lang/rusty #984: Can the compiler produce LLVM-IR

rohanrayan, 2023-10-11:
Hello 

I have a quick question. Is it possible to generate LLVM-IR from the input PLC file? It is not clear in your documentation.

Thanks.

mhasel, 2023-10-11:
Hi @rohanrayan

Thank you for your interest in our project!

You can compile to IR by passing the `--ir` build command, e.g. `plc --ir src/*.st`.
For more information about how to get started using ruSTy, check out this section of our [documentation](https://plc-lang.github.io/rusty/using_rusty.html). Also feel free to ask questions if you get stuck and feedback is always welcome.

rohanrayan, 2023-10-12:
@mhasel Thanks for your response. I am still trying to install the software and I am still having difficulties doing so. Any help in issue: https://github.com/PLC-lang/rusty/issues/985 will be appreciated!

Thanks in advance.

closed 2023-10-12 by rohanrayan