PLC-lang/rusty #985: Installation issues

rohanrayan, 2023-10-11:
**Describe the bug**
I find the installation documentation outdated. I have trouble on both windows and linux and have not been able to install the compiler on any machine upto this point.

**To Reproduce**
On Windows:
I have the following environment
cargo 1.73.0
[LLVM 14.0.6](https://github.com/PLC-lang/llvm-package-windows/releases/tag/v14.0.6) I have added /bin to the path
Windows 10 SDK & MSVC v143 - VS 2022 C++ x64/x86 build tools

I get a linker error:
lld-link: error: undefined symbol: __std_count_trivial_8
The same error as seen in https://github.com/PLC-lang/rusty/issues/955

On linux:
I tried to use the docker file, but I get this error: 

 => ERROR [internal] load metadata for ghcr.io/plc-lang/rust-llvm:latest                                                                                                        20.0s
------
 > [internal] load metadata for ghcr.io/plc-lang/rust-llvm:latest:
------
ERROR: failed to solve: rpc error: code = Unknown desc = failed to solve with frontend dockerfile.v0: failed to create LLB definition: failed to do request: Head "https://ghcr.io/v2/plc-lang/rust-llvm/manifests/latest": dial tcp: lookup ghcr.io on 8.8.8.8:53: read udp 10.47.63.250:59549->8.8.8.8:53: i/o timeout

I also tried to compile directly using cargo build, but I get this error:
cargo:warning=wrapper/lld-c.cpp:1:10: fatal error: lld/Common/CommonLinkerContext.h: No such file or directory

I have installed LLVM-14.x directly from the source in this case.

Any updates on your documentation or more verbose installation steps will be highly appreciated!

Thanks!

ghaith, 2023-10-12:
Hello
Thank you for the comment.
We recently tried the docs step by step on windows and we didn't notice the issues so it seems we are missing something here. Maybe a step we considered trivial is missing.
I would recommend trying to follow the steps that happen when building the docker container for windows. This container builds the project from scratch daily on our internal servers so it should be correct.
https://github.com/PLC-lang/rust-llvm-images/blob/main/windows/Dockerfile

Regarding the Linux docker issue this seems to be a network issue. Can you try to go to that address? The docker file can also be built locally from the same link I shared.

If you figure out what we missed please open a PR or just comment here so we add it.

mhasel, 2023-10-13:
I have tested the windows build on a local machine and I was getting linking errors aswell when triying to build with MSVC v143. After installing MSVC build tools v142, I was able to build the project without issue.
Can you try to install the v142 (you can keep your v143 installation alongside it) and report back if that resolves your issue?

rohanrayan, 2023-10-18:
Thanks. This fixes the issue

closed 2023-10-18 by rohanrayan