PLC-lang/rusty #986: Adopt new XML-schema according to IEC61131-10

mhasel, 2023-10-16:
Our current CFC/XML implementation is based on the tc6-version of the 61131-3 standard, which is largely incompatible with the newer 61131-10 version. We should look into refactoring our implementation to adhere to the newer standard and offer a conversion to allow users to port their existing projects. This would ideally be done in the IDE, independent of the compiler.

DoubleCouponDay, 2026-08-28:
@mhasel @volsa Please this Pull Request: #1896

volsa, 2026-09-29:
The CFC frontend was rewritten on the IEC 61131-10 XML format: 1e3bb857136 (#1807) removed the tc6-based plc_xml crate and 609025ec27d (#1808) added the new plc_cfc crate, whose fixtures use the `www.iec.ch/public/TC65SC65BWG7TF10` namespace. Conversion of old projects stays with the IDE, as the issue proposes, so closing this.

closed 2026-09-29 by volsa