PLC-lang/rusty #989: Does rusty support namespaces

rohanrayan, 2023-10-18:
I am trying to compile a PLC code which has [Global ](https://infosys.beckhoff.com/english.php?content=../content/1033/tc3_plc_intro/2529481355.html&id=) and [library ](https://infosys.beckhoff.com/english.php?content=../content/1033/tc3_plc_intro/2529481355.html&id=)namespaces

Could you please let me know if this is supported in RuSTy? Also could you give me small example of this if possible? For example if I have a file with globals (VAR_GLOBAL) how would I access it inside another structured text file with a namespace (GVL for example)

Thanks in advance!

ghaith, 2023-10-22:
Hello @rohanrayan , 

It's a planned feature as it is part of the standard but we still didn't get to it.
It will be part of the object oriented support I assume which we only have partly working.

rohanrayan, 2023-10-23:
Is there any timeline when I can expect this feature to be available?

ghaith, 2023-10-23:
We have no exact timeline, the plan is however currently as follows:

- Finalize CFC (Some features still missing, some library functions and debugging)
- Rework error reporting for better messages (Might also involve more documentation work)
- Direct hardware access support (%IX1.1)
- OOP

I would expect it somewhere within next year on the current plan, but we welcome contributions so I would gladly review PRs if someone contributes this.