PLC-lang/rusty #993: Action block in CFC error on VAR connection same TYPE

rarris, 2023-10-19:
we have the the following CFC code programm 
```
PROGRAM program_0
VAR_INPUT
in1,in2,in3,in4 : INT;
END_VAR
VAR_OUTPUT
out1,out2,out3 : DINT;
END_VAR
VAR
x1, x2 : INT;
END_VAR

```
in the diagram if we connect x1 to the action block input "in1" (or output), we have the follwoing compile error

`: error: Invalid assignment: cannot assign 'program_0.action1' to 'INT'`

rarris, 2023-11-22:
same error for FB->output

volsa, 2026-09-29:
The rewritten CFC transpiler (7f4c490931e, #1811) lowers action and FB block outputs to reads on the owning instance. Both cases from this issue (an action output wired to a local VAR, and an FB output wired to a VAR_OUTPUT) now compile and give the right values on master.

closed 2026-09-29 by volsa