PLC-lang/rusty #1002: Snapshot testing for resolver tests

volsa, 2023-11-02:
**Is your refactor request related to a problem? Please describe.**
I think our resolver unit tests could benefit from snapshot testing, as writing them currently is a bit tedious imo. This is somewhat related to https://github.com/PLC-lang/rusty/issues/1001.

**Describe the solution you'd like**
Introduce a function similar to `parse_and_validate_buffered` but for annotations.

riederm, 2023-11-09:
What would be a nice way of writing these tests?
For me the hard part is, to get a hold of the AST-Elements that you want to check :/

so like unpacking ´b´ in the example below is tidious
```st
foo(a := b); 
```

volsa, 2023-11-09:
I was thinking of integrating this closely with https://github.com/PLC-lang/rusty/issues/1001, so some function that generates the following output?
```
`foo(a := b)` => Function { ... }                   @ ReferenceExpr { kind: Member(...), parameters: ..., ... }
`foo(a := b)` => Value { "DINT" }                   @ CallStatement
`foo`         => Function { ... }                   @ Identifier
`a`           => Variable { "DINT", "foo.a",  ... } @ ReferenceExpr { ... }
`b`           => Variable { "DINT", "main.b", ... } @ ReferenceExpr ( ... }
```
I'm not sure if this is the best way to go, since this will basically yield all elements in the `type_map` and `type_map_hint` within the snapshot but it would make writing these tests easier in the future at the cost of _maybe_ bloated up snapshots? Maybe because in theory most resolver tests should not have a lot of elements in these `type_map(_hint)` members since they're kept small? This probably needs more discussion in the team if we really want and/or need this.

volsa, 2026-09-29:
6c860b9a702 (#1543) added `test_resolve` in `src/resolver/tests/resolver_test_util.rs`. Resolver tests can mark expressions with `{}` and snapshot their resolved type and type hint, which is the annotation helper this issue asks for.

closed 2026-09-29 by volsa