PLC-lang/rusty #1003: `--ast` flag

volsa, 2023-11-02:
**Is your feature request related to a problem? Please describe.**
For debugging purposes it would be nice to have an `--ast` flag.

**Describe the solution you'd like**
`cargo r -- <input file> --ast` should print the AST for the given input file(s)

nturley-copia, 2024-07-07:
Would you expect compilation to terminate after the AST is printed, or print the AST and then continue onto validation and codegen?

Also, what format would you expect it to print in?

ghaith, 2024-07-07:
Hello @nturley-copia,

Thank you for the message, are you willing to work on the issue? 

> Would you expect compilation to terminate after the AST is printed, or print the AST and then continue onto validation and codegen?

i would expect the compilation to stop just after the AST. @volsa? Maybe this is something to also integrate in the "config" subcommand but as well as the flag so it can generate ast on full projects.
> 
> Also, what format would you expect it to print in?

I think a simple listing, similar to what the dbg! flag does would be enough.. eventually it might be reasonable to think of serializing this, but I would go with the easy solution at first. A listing of ast elements as parsed from each POU similar to what a dbg! or a to_string does.

nturley-copia, 2024-07-07:
Yeah, I'll see if I can. I'd really like to get involved in the project, but I haven't worked with very much Rust yet so it will take a bit to get up to speed.

closed 2024-07-09 by volsa