PLC-lang/rusty #1008: Error running example simple_program

qnikil7, 2023-11-08:
I built rusty using docker on wsl, and I'm able to run the hello_world.st example.

But when I run other examples, i get errors.

- `target/debug/plc examples/simple_program.st -o simple_program --linker=cc` gives the error

  ```
  /usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/12/../../../x86_64-linux-gnu/Scrt1.o: in function `_start':
  (.text+0x17): undefined reference to `main'
  collect2: error: ld returned 1 exit status
  error: An error occured during linking
  
  Codegen Error: Compilation aborted due to previous errors
  ```
I built it again, but its the same error.

ghaith, 2023-11-09:
Hello @qnikil7 
Simple program does not have a `main` which is needed if you want to create an application. 
Try `examlpes/hello_world.st` instead.

qnikil7, 2023-11-09:
@ghaith thank you! examlpes/hello_world.st works.

closed 2023-11-09 by qnikil7

ysf-gd, 2025-07-02:
hello,qnikil7. What should i do if i want to run examples/simple_program.st without error,did you resolve it?