PLC-lang/rusty #1021: Trailing comma in array elements

volsa, 2023-11-21:
**Is your feature request related to a problem? Please describe.**
A trailing comma in array elements will generate "weird" error messages. For example `arr : [...] := [1, 2, 3, 4, 5,];` currently returns
```
error: Unexpected token: expected Literal but found ]
  ┌─ target/demo.st:3:53
  │
3 │         arr : ARRAY[1..5] OF DINT := [1, 2, 3, 4, 5,];
  │                                                     ^ Unexpected token: expected Literal but found ]

error: Unresolved constant 'arr' variable: Cannot resolve constant: EmptyStatement
  ┌─ target/demo.st:3:38
  │
3 │         arr : ARRAY[1..5] OF DINT := [1, 2, 3, 4, 5,];
  │                                      ^^^^^^^^^^^^^^^^ Unresolved constant 'arr' variable: Cannot resolve constant: EmptyStatement
``` 

**Describe the solution you'd like**
Either allow trailing commas returning a warning about it or make it a hard error.

ghaith, 2023-11-21:
maybe we just allow it in general. I like the rust style where the last comma can be ignored

teksturi, 2023-11-23:
Ok what about situation

`int_array : ARRAY[0..2, 0..2, 0..2] OF INT := [0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,];`

Should last comma still be ignored?