PLC-lang/rusty #1052: MIT License over GPL

Chaser, 2023-12-13:
Hi, this concept is super fascinating. However GPL v3 license is not so favourable. 

Would you consider changing to MIT? What would be some concerns around doing so?

Let me know if you would like to discuss more. 

Thanks, Chase.

ghaith, 2023-12-13:
Hello @Chaser,

first of thank you for checking out the project. To clarify the licensing, it is actually L-GPL V3 for the most part. The L-GPL is an extension over the GPL which is why the GPL is there. This means you could integrate it into other projects without having to make them GPL in turn.

The reason we didn't go with MIT is that we would like to get the contributions back, and I feel something like MIT or BSD is too permissive. Apache 2.0 was an option since it requires attribution, but it does not required the users to contribute back. The options were then MPL and EPL2.0 for more flexible copyleft but we eventually went with GPL. We are using GPL-3 and not GPL-2 for compatibility with the Apache 2.0 License.

I feel the LGPL for a compiler is a good balance, you can still make modifications or add features that are not GPL by simply providing the object files of the LGPL parts, which is why most of the project is created as a library and not a main. 
This means you can have the compiler be a part of a bigger toolchain, and even have support for new languages in closed source parts. The only restriction is that fixing issues in the compiler or extending the codegen/ast for new features could be reused upstream. 

I don't want to exclude a license change in the future but it might also not be easy as every contributor has to agree to it. But if there's a more flexible license that can guarantee contributions I would be open to discuss that.

riederm, 2023-12-14:
hi @Chaser, 
can you maybe elaborate where exactly you would expect some problems with L-GPL-3?

I would also like to move this into a discussion, rather than keeping this as an issue - so if nobody vetos I'll convert this into a discussion later today.