PLC-lang/rusty #1053: Codegen/compilation fails when variable length arrays are delared but not used

kumbayo, 2023-12-13:
**Describe the bug**
CodeGen fails when variable length array types are declared but not used in the program.

**To Reproduce**
Save the following text to `example.st`

```
TYPE
 ExampleVarLengthArray : ARRAY  [*] OF DINT;
END_TYPE

FUNCTION_BLOCK MyFunctionBlock
VAR_IN_OUT
    //arr: ExampleVarLengthArray;
END_VAR
END_FUNCTION_BLOCK
```

Run `cargo run example.st`
```
$ cargo run example.st 
    Finished dev [unoptimized + debuginfo] target(s) in 0.12s
     Running `target/debug/plc example.st`
error: Unknown type: __ptr_to___arr_vla_1_dint

Codegen Error: Compilation aborted due to previous errors
```
Remove the `//` in front of `arr: ExampleVarLengthArray;`
The code now compiles without error

**Expected behavior**
Code should compiles without error even when the type is not used.

**Additional context**
This problem first appeared with the commit
28e89cca644caeccb79dc136413f1e61fce87c47 Parallel compilation support (#829)

Since then, in `generate_data_types()`, `types` does not contain an entry for `__arr_vla_1_examplestruct` anymore.
When the VLA is used, the type ` __ptr_to___arr_vla_1_dint` is registered by an entry in `dependencies`

volsa, 2023-12-14:
Hi, this is _probably_ a low priority issue for us right now but since you've already investigated the issue quite a bit we would be happy to assist if you're interested in solving it :)

kumbayo, 2023-12-17:
Hi,
yes i am interested in solving it.
I already have a working solution but might take some more time until i can properly submit it.