PLC-lang/rusty #1069: Relative library pathes are borken

ghaith, 2024-01-16:
**Describe the bug**
When trying to reference a library that is relevant to the project, we always fail with a message indicating that the library could not be found.

mhasel, 2024-01-16:
I'm guessing you are referencing a build without a conf-file, where the build fails if `LD_LIBRARY_PATH` is not `.`. 

Maybe we can look into using the `$ORIGIN` [keyword](https://docs.oracle.com/cd/E19957-01/806-0641/appendixc-16/index.html) to allow dynamic, path-relative runtime linking instead of having to specify it with `LD_LIBRARY_PATH`.

ghaith, 2024-01-16:
That's also an issue, but I meant in a build config, the library includes are not found if they are relative.

ghaith, 2024-02-12:
On further investigation, it seems the -L option does nothing currently

closed 2024-02-21 by ghaith