PLC-lang/rusty #1081: Error in Standard functions regarding pointer assignments

ghaith, 2024-02-01:
> > @mhasel I think this is what you fixed in #1072 if i'm not mistaken? @rohanrayan The problem here is we are generating debug information for functions that don't exist. In a normal binary run (not IR) this does not matter because LLVM just removes it. But for IR it gets validated. The fix in #1072 should fix that I believe.
> 
> Yes, this issue should be fixed in #1072.

I checked and this commit indeed fixes this issue.

However I find other issues now. Specifically when trying to compile "string_conversion.st", I get the following error:
```
error: Invalid assignment: cannot assign '__POINTER_TO_CHAR' to 'REF_TO STRING'
    ┌─ /home/rusty_test/rusty/libs/stdlib/iec61131-st/string_conversion.st:118:5
    │
118 │     ptr := &in;
    │     ^^^^^^^^^^ Invalid assignment: cannot assign '__POINTER_TO_CHAR' to 'REF_TO STRING'
```

Also, please let me know if you want me to create new issues as this thread is getting long.

Thanks

_Originally posted by @rohanrayan in https://github.com/PLC-lang/rusty/issues/1065#issuecomment-1920895509_

ghaith, 2024-02-01:
I created a new issue for this, the problem is because of our change with diagnostics. These issues will eventually be configurable so that we could ignore them. But this specific case is a false positive.

volsa, 2026-09-29:
On current master `plc -g --ir libs/stdlib/iec61131-st/*.st` succeeds with only E067 downcast warnings and no errors, and the `ptr := &in` pattern is no longer in the stdlib sources. I could not identify one fixing commit. Closing; please reopen if you still see the error.

closed 2026-09-29 by volsa