PLC-lang/rusty #1083: FUNCTION without return type

rarris, 2024-02-05:
according to the norm, a FUNCTION without return type must be accepted!

currently we generate an error if the return type is missing.

example:
```
FUNCTION function_0
VAR_INPUT
x:INT;
END_VAR
VAR_OUTPUT
y:INT;
END_VAR
VAR
END_VAR
;
END_FUNCTION
```

console result:
```
error: Function Return type missing
Error: E001: Compilation aborted due to critical errors
```

tisis2, 2024-02-05:
why would you warn about a void function? function can be written to change the state of an application by intention and would be anoying to get warnings if you have functions like that by intention. whats the point to warn about?

ghaith, 2024-02-07:
I don't mind not having a warning. It was just because it previously was an error and the standard does not mention it explicitly

closed 2024-03-07 by volsa