PLC-lang/rusty #1090: semicolon after END_STRUCT leads to error

rarris, 2024-02-12:
```
TYPE
	S1: STRUCT
	SC2: REAL;
END_STRUCT;
	A1: ARRAY[1..99] OF INT;
END_TYPE 
```
: error: Missing expected Token KeywordColon
error: Unexpected token: expected KeywordSemicolon but found ': ARRAY[1..99] OF INT'
error: Unknown type: A1
Error: E071: Compilation aborted due to previous errors

**expected result**
according to the Norm - Semicolon after END_STRUCT must be accepted, along with multiple STRUCTs in one TYPE!

ghaith, 2024-02-12:
This is more a feature, multiple type definition in one type block, i see it as low priority because we already support multiple types in the same file, but we should do it for completion.

volsa, 2024-02-15:
Just to have this also documented on GitHub, it seems multiple type definitions within one `TYPE` block works but the semicolon is the main issue here. Specifically 
```
TYPE
    S1: STRUCT
        SC2: REAL;
    END_STRUCT

    S2: STRUCT
        SC2: REAL;
    END_STRUCT
END_TYPE
```
will generate 
```LLVM
%S1 = type { float }
%S2 = type { float }

@__S1__init = unnamed_addr constant %S1 zeroinitializer
@__S2__init = unnamed_addr constant %S2 zeroinitializer
```
Howoever adding a semicolon to `END_STRUCT` yields a syntax errors and will consequentially not generate any IR

closed 2024-02-27 by volsa