PLC-lang/rusty #1093: initialization of references support

rarris, 2024-02-13:
**Example**:
```
PROGRAM mainProg
VAR
	myInt: INT;
	myRefInt: REF_TO INT := REF(myInt);
	myF1: F1;
	myRefF1: REF_TO F1 := REF(myF1);
END_VAR
END_PROGRAM

FUNCTION_BLOCK F1
VAR_INPUT
END_VAR
VAR_OUTPUT
END_VAR
VAR
END_VAR
;
END_FUNCTION_BLOCK
```

after compilation, the following error outputs:
```
error: Unresolved constant `myRefInt` variable: Cannot resolve constant: CallStatement {
    operator: ReferenceExpr {
        kind: Member(
            Identifier {
                name: "REF",
            },
        ),
        base: None,
    },
    parameters: Some(
        ReferenceExpr {
            kind: Member(
                Identifier {
                    name: "myInt",
                },
            ),
            base: None,
        },
    ),
}
error: Unresolved constant `myRefF1` variable: Cannot resolve constant: CallStatement {
    operator: ReferenceExpr {
        kind: Member(
            Identifier {
                name: "REF",
            },
        ),
        base: None,
    },
    parameters: Some(
        ReferenceExpr {
            kind: Member(
                Identifier {
                    name: "myF1",
                },
            ),
            base: None,
        },
    ),
}
Error: E001: Compilation aborted due to critical errors
```

rarris, 2024-10-10:
error update:

`: error[E071]: Could not resolve reference to myInt Compilation aborted due to previous errors`

mhasel, 2024-10-14:
This is already tracked in #1286 and happens because referencing local, unqualified members in `REF`/`ADR` is not yet supported.