PLC-lang/rusty #1101: RTC function block

rarris, 2024-02-16:
RTC Function Block must be implemented
The other Functions from the set (TOF, TON, TP) are already implemented.

ghaith, 2024-02-16:
I cannot find RTC referenced in the standard, what section is it in?

ghaith, 2024-02-16:
References: 
[Bekhoff](https://infosys.beckhoff.com/english.php?content=../content/1033/tcplclib_tc2_utilities/35054731.html)
[Schneider](https://product-help.schneider-electric.com/Machine%20Expert/V1.1/en/standard/topics/rtc.htm)
[Logicals (neuron automation)](https://help.logicals.com/lco3docu/latest/user-documentation/en/referenzdokumentation/systembausteine-fuer-die-anwendung/iec-bausteine-fuer-die-anwendung/timerenh-funktionsbausteine/rtc-baustein)

rarris, 2024-02-16:
yea, yu're right, is not in the german Norm
but i found it in the international Norm

ghaith, 2024-02-16:
> yea, yu're right, is not in the german Norm but i found it in the international Norm

Just looked it up, it's in the old norm (second editon) but not in the 3rd edition. We should still add it for compatibility with codesys