PLC-lang/rusty #1111: Inline Enum variants with the same name do not resolve correctly

riederm, 2024-02-29:
When initializing an inline enum using one variant of that enum, the x will be resolved to one of the two. I think it is fair to say that such a line always wants x of the respective enum-type.

```st
TYPE StateA: (Idle := 0); END_TYPE
TYPE StateB : (Idle := 0, Running := 1); END_TYPE

FUNCTION main : DINT
  VAR
      variant1 : (x := 1, y := 2) := x;
      variant2 : (x := 2, y := 2) := x;
  END_VAR
  
END_FUNCTION
```

tisis2, 2024-03-01:
in IEC61131-3 6.4.4.2.1 it is mentioned that it is an error if the literal has not enought information to resolve to a unique value.
in my understanding this can be interpreted that this part should lead to an error
```
VAR
      variant1 : (x := 1, y := 2) := x;
      variant2 : (x := 2, y := 2) := x;
END_VAR
```
since you cannot address x in a uniqe way in the body of the POU

if the enum is a declared type like the StateA and StateB above you are allowed to do ```StateA#Idle``` to identify which Idle state you are using

riederm, 2024-03-01:
alright then this needs to become a validation. I got the impression that it was not reported as a problem, but i'll re-check.

riederm, 2024-03-01:
already covered by #1091

closed 2024-03-01 by riederm