PLC-lang/rusty #1119: change ENUM error to warning or ignore it

rarris, 2024-03-05:
in the following example we have a function call on the right side of enum variants and this is wrong unless the function returns an enum

```
PROGRAM mainProg
VAR
toFB : function_block_0;
toDT : struct_io;
END_VAR
toFB(i_enum := toDT.val3);
END_PROGRAM
```
```
FUNCTION_BLOCK function_block_0
VAR_INPUT
i_enum : enum1;
END_VAR
VAR_OUTPUT
END_VAR
VAR
END_VAR
;
END_FUNCTION_BLOCK
```
```
TYPE struct_io :
	 	STRUCT 
		val1 : BOOL;
		val2 : UINT;  
		val3 : SINT;
	END_STRUCT
END_TYPE
```
```
TYPE enum1 : (
		element0,
		element1
	) ;
END_TYPE
```
**_mainProg.st:7:16:{7:16-7:25}: error: Value evaluated at run-time, use an enum variant from `enum1`
Error: E001: Compilation aborted due to critical errors_**

after some disscusions with @volsa and @ghaith we should change this type of error to warning or even ignore it

volsa, 2024-03-05:
I'm still not a big fan of converting it to a warning, especially because the generate code can become unsound but after discussing it with Ghaith I'm fine if we do end up with a warning instead of an error here.

ghaith, 2024-03-06:
The interesting thing is I checked this with @tisis2 in the norm yesterday and they even have it in an example. They assign an out of range feature without and say this is no error. They do it however only on a named values type, which are not exactly the same as enums according to the norm. 
Just wanted to mention it as in a warning is better than an error because the norm even allows it. But we could eventually have a separation between normal enum and named values where one would be an error and one would be allowed.

tisis2, 2024-03-06:
exactly thats basically the behavior you could understand from the norm.
on the enum type in 6.4.4.2.1 it is mentioned each dataelement of that type can only take a value that is mentioned in the value list.
for the type with assigned values it is mentioned in 6.4.4.3.1 that the declaration of the values does not limit the value range of variables of this type, so you still cann assign other constants or expressions.
the example in the same chapter even shows a case where an assignment of a expression is allowed for this kind of type but mentions that this is not allowed for a simple enum

closed 2024-03-06 by volsa