PLC-lang/rusty #1121: Testing with `--release` leads to a segfault

mhasel, 2024-03-05:
**Describe the bug**
Running our tests with the release flag results in a segfault:
```
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00005604d57c9114 in core::hint::black_box<core::result::Result<(), alloc::string::String>> () at library/core/src/hint.rs:286
286     library/core/src/hint.rs: No such file or directory.
```

When looking further up the stack-trace, this seems to be threading-related:
```
#17 std::thread::{impl#0}::spawn_unchecked_::{closure#1}<test::run_test::{closure_env#1}, ()> () at library/std/src/thread/mod.rs:528
#18 core::ops::function::FnOnce::call_once<std::thread::{impl#0}::spawn_unchecked_::{closure_env#1}<test::run_test::{closure_env#1}, ()>, ()> () at library/core/src/ops/function.rs:250
#19 0x00005604d8b45125 in alloc::boxed::{impl#47}::call_once<(), dyn core::ops::function::FnOnce<(), Output=()>, alloc::alloc::Global> ()
    at library/alloc/src/boxed.rs:2007
#20 alloc::boxed::{impl#47}::call_once<(), alloc::boxed::Box<dyn core::ops::function::FnOnce<(), Output=()>, alloc::alloc::Global>, alloc::alloc::Global> () at library/alloc/src/boxed.rs:2007
#21 std::sys::unix::thread::{impl#2}::new::thread_start () at library/std/src/sys/unix/thread.rs:108
#22 0x00007fd850a94ac3 in start_thread (arg=<optimised out>) at ./nptl/pthread_create.c:442
#23 0x00007fd850b26850 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
```
Unfortunately, the trace stops here.

**To Reproduce**
Steps to reproduce the behavior:
Run `cargo t --release`

ghaith, 2024-03-06:
And the report when not adding workspace:
```
      PASS [   0.019s] rusty::tests correctness::math_operators::substraction::substract_array_basic
------------
     Summary [   4.216s] 1849/1976 tests run: 1845 passed, 4 failed, 15 skipped
     SIGSEGV [   0.377s] rusty::tests correctness::functions::mux_string_literal
     SIGSEGV [   0.381s] rusty::tests correctness::functions::mux_string_ref
     SIGSEGV [   0.369s] rusty::tests correctness::functions::sel_string_literal
     SIGSEGV [   0.363s] rusty::tests correctness::functions::sel_string_ref
error: test run failed
```

mhasel, 2024-03-06:
These tests also segfault in debug builds when adding  
```
[profile.dev]
opt-level = 1
```
to `Cargo.toml`. This should be helpful for debugging this issue (`--release` defaults to `opt-level = 3`). `

PR #1142 (2024-03-11): fix: string tests no longer segfault when testing with optimization
Some of our string tests caused segfaults when running tests with either `--release` or adding a `[profile.test]` with an `opt-level = 1` or higher to `Cargo.toml`.
This was caused to due not reserving enough memory for test-results in our test code. Initially I wanted to commit this along with changes to the CI to add release tests, but since there are quite a few more tests causing issues, I have decided to split the fixes into their own commits.

To verify that no segfaults occur in our tests, you can
`cargo -t --release -- [... cut]
closed 2024-03-11 by mhasel