PLC-lang/rusty #1122: Add `--release` tests to our CI

mhasel, 2024-03-05:
We should add release testing to our CI to get ahead of future issues regarding release builds.

mhasel, 2024-03-09:
Currently blocked by

- [ ] Tests causing segfaults:
```
SIGSEGV [   0.099s] rusty::tests correctness::functions::mux_string_literal
SIGSEGV [   0.091s] rusty::tests correctness::functions::mux_string_ref
SIGSEGV [   0.060s] rusty::tests correctness::functions::sel_string_literal
SIGSEGV [   0.058s] rusty::tests correctness::functions::sel_string_ref
SIGSEGV [   0.046s] rusty::tests correctness::strings::string_returned_from_generic_wrapper_function_does_not_truncate
SIGSEGV [   0.125s] iec61131std::string_conversion_tests string_to_wstring_conversion
SIGSEGV [   0.123s] iec61131std::string_conversion_tests string_to_wstring_extra_conversion
SIGSEGV [   0.112s] iec61131std::string_conversion_tests wstring_to_string_conversion
SIGSEGV [   0.110s] iec61131std::string_conversion_tests wstring_to_string_extra_conversion
SIGSEGV [   0.110s] iec61131std::string_function_tests delete_ext_string_with_escape_sequence
SIGSEGV [   0.087s] iec61131std::string_function_tests delete_string_with_escape_sequence
SIGSEGV [   0.094s] iec61131std::string_function_tests left_wstring
```
tracked in #1121 

- [ ] Bitaccess returning incorrect values for booleans
```
FAIL [   0.035s] iec61131std::bit_conversion_tests byte_to_bool
FAIL [   0.035s] iec61131std::bit_conversion_tests dword_to_bool
FAIL [   0.028s] iec61131std::bit_conversion_tests lword_to_bool
FAIL [   0.034s] iec61131std::bit_conversion_tests word_to_bool
```
tracked in #1141 

- [ ] `date_time_extra_functions::concat_dt` panicking due to hitting `.expect("Invalid parameters, cannot create TOD")` in function body (`in1` shows garbage values in the debugger)
```
FAIL [   0.045s] iec61131std::date_time_extra_functions_tests concat_dt_signed_ints
FAIL [   0.039s] iec61131std::date_time_extra_functions_tests concat_dt_unsigned_ints
FAIL [   0.056s] iec61131std::date_time_extra_functions_tests concat_ldt_signed_ints
FAIL [   0.042s] iec61131std::date_time_extra_functions_tests concat_ldt_unsigned_ints
```

- [ ] `date_time_numeric_functions::checked_mul_time_with_signed_int` returning garbage values
```
FAIL [   0.025s] iec61131std::date_time_numeric_functions_tests mul_ltime_signed
FAIL [   0.032s] iec61131std::date_time_numeric_functions_tests mul_signed
FAIL [   0.028s] iec61131std::date_time_numeric_functions_tests mul_time_signed
```

- [ ] result of  `validation_functions::is_valid_bcd` failing expected assertion (`assert!(maintype.valid);`)
`iec61131std::validation_functions_tests is_valid_word`

volsa, 2026-09-29:
CI now runs release builds: 52c21a615c1 (#1708) added the `test-linux-release` job to `linux.yml` (and fixed the signed `mul_*` blockers), and `lit.yml` has a `lit-linux-release` job. Closing.

closed 2026-09-29 by volsa