PLC-lang/rusty #1127: Performance got worse in the last 2 weeks

ghaith, 2024-03-06:
It might be the change with the parser errors, but I noticed that on big projects a lot of time would pass before we see the first error. We should investigate that.

volsa, 2024-03-11:
Did some quick research and it looks like I have unlocked an achievement, unintentionally fucking up the performance by a lot due to https://github.com/PLC-lang/rusty/commit/94a74fd429b0e3da03e795e181342823fa402615. Specifically the [`find_local_member`](https://github.com/PLC-lang/rusty/blob/e39e2421cc81d3c0ae18684b8894ccb38fd05167/src/index.rs#L1059-L1072) method internally calls [`find_enum_variant_in_pou`](https://github.com/PLC-lang/rusty/blob/e39e2421cc81d3c0ae18684b8894ccb38fd05167/src/index.rs#L1184-L1189) which works on raw (sub)string comparisons. I knew it would have some performance penalties (especially because there's no caching for already performed queries) but I didn't think it would be this bad. Anyways, seems like we need to refactor how enums are stored in the index hence closely related to https://github.com/PLC-lang/rusty/issues/1112. [image]

volsa, 2024-03-11:
Also https://plc-lang.github.io/metrics/ wasn't able to detect any performance regression, we really need to find some big open source library which we can include in the metrics...

closed 2024-04-04 by volsa