PLC-lang/rusty #1164: function block call - "output is not a STRING"

rarris, 2024-03-19:
```
PROGRAM mainProg
VAR
   fb1 : function_block_0;
END_VAR
   fb1(
     input1 := IN1,
     output => OUT1
   );
END_PROGRAM

FUNCTION_BLOCK function_block_0
VAR_INPUT
  input1 : inDATA;
END_VAR
VAR_OUTPUT
  output : outDATA;
END_VAR
VAR
END_VAR
;
END_FUNCTION_BLOCK

TYPE inDATA :
 STRUCT
	  Id:UDINT;
 END_STRUCT
END_TYPE

TYPE outDATA :
 STRUCT
  	  Parm:UDINT;
 END_STRUCT
END_TYPE

VAR_GLOBAL
  IN1 : inDATA;
  OUT1 : outDATA;
END_VAR
```

mainProg.st:7:12:{7:12-7:16}: error[E071]: outDATA is not a String
Error: Compilation aborted due to previous errors

mhasel, 2024-03-21:
I have reduced the reproducible example down a little:
```
TYPE OUT_TYPE : STRUCT 
    a : BYTE;
END_STRUCT;
END_TYPE

FUNCTION_BLOCK FB
VAR_OUTPUT 
    output : OUT_TYPE;
END_VAR    
END_FUNCTION_BLOCK

PROGRAM PRG
VAR
    out: OUT_TYPE;
    station: FB;
END_VAR
    station(output => out);
END_PROGRAM
```

mhasel, 2024-03-25:
This happens here:
https://github.com/PLC-lang/rusty/blob/b357a05cb064eb21fed110b6d7c5f543f72c18e8/src/codegen/generators/expression_generator.rs#L574C1-L579C52

`generate_string_store` then expects a string and will fail on any other type. Simply commenting out the `OR`ed conditions for structs and arrays will get rid of the error.
Structs and arrays will then be generated via stores, so maybe related to #1074 too.
I might be missing something here - do we need special handling for arrays and structs in output assignments? The comment only mentions strings.

PR #1234 (2024-06-04): fix: aggregate output assignments
Renames `generate_string_store` to `build_memcpy` and adapts the logic to work for all aggregate types, not just strings.

Resolves #1164
closed 2024-06-04 by mhasel