PLC-lang/rusty #1167: show warning for BOOL-INT downcast

rarris, 2024-03-20:
the following code 

```
PROGRAM mainProg
VAR
  x1 : BOOL;
  x2 : INT;
END_VAR
  x1 := 20; 
  x2 := TRUE;
END_PROGRAM
```

will generate no warning, no error!

It will be nice to have minimum a warning and later to adjust easier the error_config file for codesys where this a error is.

mhasel, 2024-03-20:
I don't really see `x2 := TRUE` as a big issue, it'll just have the value 1.
Your other example will also retain it's value, so no downcast is happening. Any non-zero value will evaluate to `TRUE`.

And if you rewrite your assignment to `x1 := x2`, in other words INT to BOOL, the compiler will warn about the downcast.

The only problem I can see here is if you assign a negative number to a boolean, which will underflow `u8` and then reassign it into a signed variable.
E.g.:
```

{external}
FUNCTION printf : DINT
VAR_INPUT {ref}
  format : STRING;
END_VAR
VAR_INPUT
  args : ...;
END_VAR
END_FUNCTION

FUNCTION main : DINT 
VAR
  x1 : BOOL;
  x2 : INT;
END_VAR
  x1 := -2; // will underflow u8 and have value 253

  IF x1 THEN
    printf('evaluates to true, has value %d$N', x1);
  END_IF;

  x2 := x1; // will still have value 253
  printf('value is %d', x2);
END_FUNCTION
```

So maybe we could warn about the implicit conversion to unsigned when assigning a signed literal to a boolean.

volsa, 2024-03-20:
`x1 := 20` just feels wrong to me tbh, I'd be in favour of at least emitting a info / warning here.

mhasel, 2024-03-20:
> `x1 := 20` just feels wrong to me tbh, I'd be in favour of at least emitting a info / warning here.

This is basically the same as #1148, just in a different context/a literal in this case. But disallowing literal integers to booleans seems sensible.

volsa, 2026-09-29:
Both assignments from the example are now reported as E037 errors (`cannot assign 'DINT' to 'BOOL'` and `cannot assign 'BOOL' to 'INT'`) since 80cc6c33348 (#1924), which is stricter than the requested warning. The severity can be changed through the error config if needed.

closed 2026-09-29 by volsa