PLC-lang/rusty #1171: Unary expression `+ <REAL>` panics on `unimplemented!()`

mhasel, 2024-03-20:
**Describe the bug**
Unary expressions with an explicit `+` operator on REALs will panic in https://github.com/PLC-lang/rusty/blob/b357a05cb064eb21fed110b6d7c5f543f72c18e8/src/codegen/generators/expression_generator.rs#L400
The same expression with INTs is fine.

**To Reproduce**
This came up while porting an internal project and the MRE boils down to this:
```
FUNCTION main: DINT
VAR
    r: REAL;
    i: INT;
END_VAR
    i := + 6; // this is fine
    r := + r; // hits `unimplemented!()` in `expression_generator.rs:400`
    r := + 6.0; // same as above
END_FUNCTION
```

**Expected behavior**
the expression `+ 6.0` behaving just like `6.0`

volsa, 2024-03-21:
Related PR https://github.com/PLC-lang/rusty/pull/640

volsa, 2026-09-29:
Fixed by 9a110501971 (#1513), which handles `Operator::Plus` in unary codegen independent of the type. The example from this issue now compiles on master, and `r := + r` / `r := + 6.0` emit a plain load and store.

closed 2026-09-29 by volsa