PLC-lang/rusty #1182: Linking fails on enums with the same variant name

mhasel, 2024-03-27:
**Describe the bug**
Given you have multiple enums with overlapping/same variants across multiple files, when you try to compile them then linking will fail with 
```
error[E071]: An error occurred during linking: lld: error: duplicate symbol: <symbol_name>
```
regressed commit: 94a74fd

**To Reproduce**
Steps to reproduce the behavior:
foo.st:
```smalltalk
FUNCTION_BLOCK FOO
VAR
	foo_enum : (start, stop) := stop;
END_VAR
END_FUNCTION_BLOCK
```
bar.st:
```smalltalk
FUNCTION_BLOCK BAR
VAR
	bar_enum : (stop, go) := go;
END_VAR
END_FUNCTION_BLOCK
```
Compiling with `plc foo.st bar.st`:
```bash
error[E071]: An error occurred during linking: lld: error: duplicate symbol: stop
>>> defined at foo.st
>>>            /tmp/.tmpJFdPQo/target/foo.st.o:(stop)
>>> defined at bar.st
>>>            /tmp/.tmpJFdPQo/target/bar.st.o:(.rodata.cst4+0x4)
```

riederm, 2024-04-02:
this problem came with the refactoring when we started to parallelize codegen into separate objects. In early versions we generated everything into one object and llvm took care of this situation by appending a number to duplicate global variables:

https://github.com/PLC-lang/rusty/blob/2294494aaec1fef2a0b10e6e98308e6dcb7b8bc7/src/tests/adr/enum_adr.rs#L72-L93

We probably need to pick unique llvm-names for the global constants now.

mhasel, 2024-04-02:
> this problem came with the refactoring when we started to parallelize codegen into separate objects. In early versions we generated everything into one object and llvm took care of this situation by [...]

@ghaith and I took a look at this last week and came to the same conclusion. This might be an easy fix by mangling the global with the pou qualifier, but I have not tried it yet:
https://github.com/PLC-lang/rusty/blob/2294494aaec1fef2a0b10e6e98308e6dcb7b8bc7/src/codegen/generators/variable_generator.rs#L118-L119C102

commit 0c6a029e (2024-04-04): fix: enum-variant names no longer clash
... when codegen runs parallel.

Enum variant globals are now generated using their qualified name instead
of their variant name.

Resolves #1182
PR #1188 (2024-04-18): fix: enum-variant names no longer clash
... when codegen runs in parallel.

Enum variant globals are now generated using their qualified name instead of their variant name.

resolves #1182
commit 66ce86a5 (2024-04-18): fix: enum-variant names no longer clash (#1188)

* fix: enum-variant names no longer clash
... when codegen runs in parallel.

Enum variant globals are now generated using their qualified name instead
of their variant name.

Resolves #1182
closed 2024-04-18 by mhasel