PLC-lang/rusty #1192: Leading comma in section name when running tests under MacOS

volsa, 2024-04-05:
**Describe the bug**
Currently `cargo test` will fail with snapshot diffs because a leading comma is appended to the section names. For example the test `multiple_files_with_debug_info` will fail due to 
```diff
-define i16 @main() section "fn-main:i16" !dbg !10 {
+define i16 @main() section ",fn-main:i16" !dbg !10 {

-declare !dbg !18 void @mainProg(%mainProg*) section "fn-mainProg:v"
+declare !dbg !18 void @mainProg(%mainProg*) section ",fn-mainProg:v"

-define void @mainProg(%mainProg* %0) section "fn-mainProg:v" !dbg !10 {
+define void @mainProg(%mainProg* %0) section ",fn-mainProg:v" !dbg !10 {
```
I'm unsure if this is a MacOS thing or ARM in general but I suspect the former.

**To Reproduce**
Under MacOS 14.4.1 switch to the master branch and run `cargo t`

**Expected behavior**
The snapshots shouldn't change

volsa, 2024-04-25:
I tried fixing the snapshot tests under MacOS, removing these leading commas for all tests in the `codegen/` folder such that the IR is identical to the master branch. While its working (see https://github.com/PLC-lang/rusty/compare/macos-codegen) the correctness and integration tests fail with a sigbus panic. Not sure why, but the section names seem to cause the panic under MacOS. This needs further investigation...

teksturi, 2024-08-29:
https://github.com/TheDan64/inkwell/pull/398

here is little bit more info for this one when someone needs to look into this.

closed 2026-02-27 by volsa